Severson-Group/AMDC-Firmware #10: Create AMDC board definition file for Vivado

npetersen2, 2019-05-25:
Currently, the Vivado project is based around the `em.avnet.com:picozed_7030_fmc2:part0:1.1` board definition from online.

Please create a clone of this file, but tailor it to AMDC. At a minimum, include a nice picture and change the name. I think this board definition file can specify ports, etc... Look into this.

BrandtBuchda, 2021-05-17:
The `.zip` of board definitions from [Avnet](https://www.avnet.com/wps/portal/us/products/avnet-boards/avnet-board-families/microzed/) no longer contains the `em.avnet.com:picozed_7030_fmc2:part0:1.1` board definition we use in Vivado. I had to take the board definition from an old computer when I installed Vivado on a new one. It might be a good idea to put that board definition in a repo somewhere until this issue is resolved.

npetersen2, 2021-05-25:
For reference, attached is the zip file of the correct board file
[picozed_7030_fmc2.zip](https://github.com/Severson-Group/AMDC-Firmware/files/6540700/picozed_7030_fmc2.zip)

elsevers, 2026-06-03:
Nice @AwesomeTornado!

AwesomeTornado, 2026-06-03:
@elsevers 
I've updated the organization, display name, and part name, and modified the build script to use our custom board.

I do not know if there is anything else we might want to change. Nathan was right that we can change things like ports, default IP blocks, etc, but I'm not sure what we might want to change here.