Severson-Group/AMDC-Firmware #9: Get Ethernet to work

npetersen2, 2019-05-25:
We need to get Ethernet to work on the new AMDC boards. John Compas got it working, but Nathan Petersen cannot replicate his work.

Please either get a simple echo server working over Ethernet, or get Eric's `basic` app working with Ethernet and the `amdc_tool`.

npetersen2, 2020-07-26:
This is still an outstanding issue, that is becoming more and more important (for debugging / streaming control variables in real-time)... I have investigated this for hours in the past; my conclusion is always that the issue is with the PicoZed PHY, not the AMDC. LwIP doesn't initialize correctly, so perhaps it's not configured correctly for the hardware platform...?

npetersen2, 2021-03-10:
@BrandtBuchda is working on this issue. He got the Ethernet to work on the AMDC REV D hardware platform.

The required change was in the Vivado settings for the ZYNQ block.

@BrandtBuchda please add a comment on this issue describing the required change. Also, please update the Vivado block diagram file and create a new PR which sets the correct settings.

BrandtBuchda, 2021-04-12:
There was a configuration issue in the ZYNQ processing system block. The physical ethernet layer was not being connected to the DSP. The MDIO bus needed to be enabled (shown below) to connect the ethernet circuits to the MAC on the DSP. 

[image]

PR #189 (2021-04-13): Configured PS for ethernet
This PR configures the processing system in the Vivado Block Diagram to enable the ethernet through the DSP.  The AMDC block diagram is the only file that changed.

Closes: #9
closed 2021-04-13 by BrandtBuchda