Severson-Group/AMDC-Firmware #36: Write GPIO driver

npetersen2, 2019-07-21:
Need to write the GPIO which sets GPIO pins high / low.

This will be both C and FPGA. C should just write to a register. Need to build FPGA peripheral which *is* the GPIO. Also, need to figure out how to do GPIO as well as DAC. Probably have single DAC IP in the FPGA and mux it to either GPIO port via config register.

npetersen2, 2019-07-22:
Note that in the FPGA, all the GPIO pins will need to be of type `inout` so that we can dynamically change their state from INPUT to OUTPUT at run-time.

https://electronics.stackexchange.com/questions/356632

npetersen2, 2020-07-26:
This is still an outstanding issue. However, it is more complicated than before since we have two supported hardware targets with different GPIO peripherals: REV D = isoSPI + diff I/O; REV C = Generic 12 pin I/O.

In general, this driver should make the GPIO of the AMDC function like the generic GPIO peripherals on regular MCUs.. i.e. STM32, EFM32, TI MCUs, etc... Users should be able to configure GPIO, add interrupts, etc.

The GPIO might be evolving in the next hardware revision, so this driver's goals might be a moving target...

npetersen2, 2023-05-04:
After a few years, updating this issue...

The REV C hardware is no longer supported, only REV D & E. These have a more unified GPIO interface.

@codecubepi has been working on a general purpose driver block for pin-level GPIO access, see #296. We have also added in a GPIO mux to handle the different IP blocks which want control over the GPIO port.

Therefore, when @codecubepi merges #296, let's close this issue.

codecubepi, 2023-05-04:
@npetersen2 

I figured I would browse the issues and found this too.

Can I link this one to my PR as well?

closed 2023-05-04 by codecubepi