Severson-Group/AMDC-Firmware #61: Parallel DPNV winding current regulator

elsevers, 2020-02-23:
**Challenges** that are arising from using a generalized current regulator for DPNV parallel windings:
1. If we use isolated suspension/torque dc bus supplies to regulate terminal currents, we end up trying to back feed power into the unidirectional suspension voltage bus. This causes voltage faults in the supply. Not good. (More reading material can be found on this problem / feature here: https://ieeexplore.ieee.org/abstract/document/8919969)
2. If we use one suspension/torque supply, we solve problem 1 but get circulating currents due to the lack of zero-sequence current regulation
3. The suspension current measurements should have 1/2 of the torque current measurement subtracted off of them (following implementation depicted in Fig. 4 and 5 of https://ieeexplore.ieee.org/abstract/document/8738835). We can fudge this manually for DC conditions by adjusting the reference currents of the suspension regulator, but for AC conditions this hack doesn't work (suspension current space vector rotates backwards).

**Suggested solution:**
Given these challenges, I suggest creating a parallel dpnv current regulator user app that piggy-backs on the capability added in #issue-57-add-generic-cc (that is, it uses the current regulators of #issue-57-add-generic-cc) by manipulating their I/O and adding additional features as described below:

1. Assumes a single supply for suspension/torque dc buses and therefore implements zero sequence current control following discussion/description in https://ieeexplore.ieee.org/document/7296654 - Eq 10 and Fig. 8. The output voltages of this regulator need to be added to the output voltages of the suspension regulator to determine the final PWM. I suggest we make it possible to enable/disable this regulator so that we can switch between isolated buses and a non-isolated bus.
2. The suspension current measurement should have 1/2 of the torque current measurement subtracted off of them (following procedure depicted in Fig. 4 and 5 of https://ieeexplore.ieee.org/abstract/document/8738835) before being compared to Ix* and Iy*.
3. The current regulator should maintain the ability to set independent Id, Iq, Ix, Iy, omega_dq, omega_xy

npetersen2, 2020-07-27:
@elsevers @ASHAD1 is this issue still relevant? Does the code @ASHAD1 is using in his user repo do the required functionality? Or do we still need to build this?

npetersen2, 2021-03-10:
@elsevers is this issue still relevant? Do we need to create this?

npetersen2, 2021-12-27:
This is an application-specific example.

Therefore, I moved it to the `AMDC-Examples` repo: https://github.com/Severson-Group/AMDC-Examples/issues/10.

closed 2021-12-27 by npetersen2