Severson-Group/AMDC-Firmware #163: Analog inputs are too noisy

npetersen2, 2020-10-08:
*Making this issue in the firmware repo, although the solution is probably in the hardware; not sure yet.*

## Description

This issue documents my findings after analyzing the ADC noise on the AMDC native analog inputs (REV D hardware). I am comparing the noise to the new external [SensorCard platform](https://github.com/Severson-Group/SensorCard) data, using the [`LVCard`](https://github.com/Severson-Group/SensorCard/tree/develop/AnalogCard) sensor cards (REV B).

## Experimental Set-Up

I am using a BK Precision 4040A function generator to output a sine wave of ~16V pk-pk (+8V to -8V). This is fed into the analog inputs of the SensorCard platform as well as the AMDC native analog inputs (for comparison). For some channels, I have inverted the + and - so that the input is flipped in sign.

The AMDC C code interface to both the SensorCard data and the native AMDC data look the same to the user: FPGA registers are populated "behind the scenes" with the latest ADC data. The data is acquired in sync with the PWM carrier on the AMDC. The latest valid data is returned to the user code.

In all tests, a task runs in an AMDC user app which pulls the latest analog sample at 10kHz and stores it for the logging engine. The logging engine samples this data at 10kHz. The plots below are the result of this logged data.

For all tests, the input sine wave is at ~270Hz.

## Test Results

In these test results:
- `adc_ch*` indicates data from the AMDC native analog inputs
- `mb_ch*` indicates data from the "motherboard" i.e. SensorCard platform

### Test 1: Raw Data

This shows the sampled data straight from the function generator. The data looks good: both the AMDC and SensorCard give about the same sine waves!

A few blimps can be seen on the second plot (see the green waveform at t = ~0.020 sec or ~0.015). This is due to sample timing issues with the SensorCard interface. This is an issue, but is not the focus of this experiment. 

[image]

### Test 2: 1st Difference of Raw Data

The above data is processed using the pandas `data.diff()` method which takes the difference between consecutive data points, effectively the discrete-time derivative.

Now, the imperfections are very visible. The AMDC data is very noisy, while the SensorCard data is very clean (except for the timing issues, duly noted). This shows that the function generator output is indeed not noisy, since the SensorCard is able to read a very clean signal. Something is wrong with the AMDC native inputs.

[image]

### Test 3: 2nd Difference of Raw Data

The original data is now processed with `data.diff().diff()`, effectively the 2nd derivative.

Clearly, it is seen that the AMDC native inputs are MUCH more noisy than the SensorCard. Interestingly, the SensorCard data shows a bizarre noise spike at the peaks.

[image]

### Test 4: Raw Data, Time Delays

The original data is now plotted with less time range, showing that the SensorCard data is delayed by one Ts (1/10000 sec). This is a known issue and is not the subject of this investigation.

[image]

### Test 5: 1st Diff of Test 4 data

The `data.diff()` output is shown for the previous test:

[image]

### Test 6: AMDC analog inputs only (noise correlation study)

A study of only the AMDC inputs is performed. Here, the same sine wave is fed into 5 of the 8 AMDC inputs. Nothing is connected to the 6-8 inputs (thus the weak tie resistor should effectively result in 0V input for these).

Below shows the results. Each subplot shows:
1. raw data (sample input to ch1..5)
2. raw data from ch6..8 (should read 0V)
3. 3rd derivative of plot 1
4. 3rd derivative of plot 2

[image]

**This is very interesting.** The 3rd plot shows the noise profile of all the AMDC inputs 1..5. CLEARLY, they exhibit the same noise! Somewhere, either in hardware or firmware, this noise is being introduced.

## Solutions

A few ideas on solutions to this noise issue:

1. Maybe, the AMDC routing has digital lines which cross over the analog input lines. This would result is coupling and thus noise. Since the ADC SPI clock is ran at 50MHz, this could introduce significant noise into the system. However, I thought when I designed the board I separated the digital and analog domains. I will post a follow-up comment showing the routing.

2. The raw input voltage is actually not noisy. In the current firmware, the FPGA drives the ADC to sample at ~800kHz and then only saves the data which correspond to the PWM carrier peak and valley. Perhaps the capacitive current draw into the ADC at 800kHz is causing noise on the inputs. They might not be buffered enough, or the charge caps might be too small. For example, the VREF cap which feed the ADC is only 100pF per analog channel negative input. Maybe, sampling at this fast rate is causing voltage ripple on those VREF caps, or the main signal caps... I will post a follow-up looking into this.

## Related People

Please give this a read and post your comments:
- @elsevers 
- @lsclari 
- @ASHAD1 
- @mimayberry 
- @Anvar-Khamitov 
- @ngadiyar93 
- @GnomeTek

npetersen2, 2020-10-08:
Per @GnomeTek request:

FFT of raw data for all 8 AMDC input channels. Data gathered over 5 seconds at 10kHz, so ~50k samples.

Inputs 1-5 have 275Hz sine input (see peak in FFT), 6-8 have not voltage input.

[image]

npetersen2, 2020-10-08:
I think the most obvious answer here is to just add a LPF in the FPGA at ~100kHz bw (user configurable)... This should help solve this!

See issue #12 for discussion on the LPF.

elsevers, 2020-10-10:
@npetersen2 Thank you for your very detailed investigation and write-up. This is very helpful for all of our review and will be a great future reference for ADC issues (as well as an excellent example of how to proper document technical issues with the AMDC). 

I like your idea of adding a programmable LPF in the FPGA fabric. 

A user should set this according to the sampling rate of their control loop to prevent aliasing. I.e., with a 10 kHz control loop, this filter's bandwidth should have a max value of 5 kHz (and preferably lower, from an aliasing perspective). 

**Summarizing the design of filters in our analog signal chain:**

`Real world signal` &rightarrow; `anti-aliasing hardware filter` &rightarrow; `ADC sampling` &rightarrow; `FPGA: anti-aliasing digital filter` &rightarrow; `AMDC control task sampling`

 - `anti-aliasing hardware filter`: bandwidth should be determined based on `ADC sampling` rate (the bandwidth should be less than half of this sampling rate)
- `FPGA: anti-aliasing digital filter` bandwidth should be determined based on `AMDC control task sampling` rate OR a lower value if the bandwidth of the controller reasonably permits it (i.e., your example of a motion input, where the motion regulator may only have a bandwidth of ~10-50 Hz)

npetersen2, 2020-10-10:
@elsevers Question about hardware anti-aliasing filter:

Our current architecture is: hardware RC filter is at ~50kHz. ADC is sampled at ~800kHz, but then downsampled to the PWM rate by just copying the latest data into the output register in sync with the PWM carrier. Thus, I believe one could say the effective sample rate is still only the PWM rate (e.g. 200kHz for 100kHz PWM output, peak and valley of triangle carrier).

All this to say, the hardware RC filter should have bandwidth of <0.5x the "effective" sampling rate (PWM-based rate).

So, for our system, with hardware RC at 50kHz bandwidth, drive PWM freq ought not go below 50kHz since this could cause aliasing (50kHz PWM implies 100kHz sampling due to double sampling per period).

elsevers, 2020-10-10:
Yes, I think you are correct on this. To go to switching frequencies below 50 kHz (and still follow these guidelines) a user should modify their hardware low pass filter by soldering in different values of R or C. 

This scenario will certainly occur for someone in the lab. For example, @GnomeTek will likely need a switching frequency of 20kHz. 

Another reason for maintaining up to date board logs for the AMDC.

npetersen2, 2020-10-22:
For future reference, this snippet of code generates the FFT plots from above (might have to adjust for different variable names from AMDC, etc).

```python
# Number of sample points
N = data.shape[0]

# Sample spacing
T = 1.0 / 10000.0

xf = np.linspace(0.0, 1.0/(2.0*T), N/2)

fig, ax = plt.subplots(8,1, figsize = (15,15))

names = ["mb_ch1", "mb_ch2", "mb_ch3", "mb_ch4", "mb_ch5", "mb_ch6", "mb_ch7", "mb_ch8"]

i = 0
for n in names:
    yf = scipy.fftpack.fft(np.nan_to_num(data[n].to_numpy()))
    ax[i].plot(xf, 2.0/N * np.abs(yf[:N//2]))
    ax[i].set_xscale('log')
    ax[i].set_yscale('log')
    ax[i].grid()
    ax[i].set_xlim([100, 5000])
    ax[i].set_ylabel(n)

    # Plot harmonic lines
    #f = 271
    #for j in [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20]:
    #    ax[i].axvline(j*f, color="r");
    
    i = i + 1

ax[7].set_xlabel("Frequency (Hz)")

plt.show()
```