Severson-Group/AMDC-Firmware #198: AMDC ICC / Ethernet Logging

BrandtBuchda, 2021-05-17:
There are still some issues with the AMDC Inter-core Communication and Ethernet Logging projects that need to be addressed before they can be merged into develop. This issue will highlight some of the problems I've encountered and also outline what I've learned along the way.

# Ethernet TCP/IP LWIP App

- The ethernet TCP/IP stack is built using the included LWIP distribution in the Xilinx SDK. The application makes use of the RAW LWIP drivers as apposed to the LWIP sockets for the increased throughput the RAW mode can achieve. In my testing the RAW mode was able to achieve data throughput of ~800Mbps on the 1Gbps ethernet port. ([Iperf3](https://iperf.fr/) was used to measure the throughput). I was running the included LWIP TCP throughput applications in the Xilinx SDK.

- Although the LWIP libraries can be included in the BSP easily, configuring the LWIP settings is not as simple. LWIP makes use of several header files throughout the library for configuring everything from switching between UDP/TCP, to the size of the buffers. Since the library is supplied through the Xilinx SDK installation all modifications to the LWIP settings have to be completed through the BSP configuration interface in the Xilinx SDK. The settings are then compiled and the libraries are automatically updated to reflect the changes. I ran in to an issue with this where certain macros in some of the header files that were not the main `lwipopts.h` wouldn't be updated when I was trying to configure LWIP to run in RAW mode. The macros can't be manually updated because they are overwritten every time Xilinx builds the project. To get around this I have been developing off the built in RAW TCP applications supplied by Xilinx where the LWIP library is configured as it needs to be. Its possible that there is some setting in Xilinx that would resolve this issue but I think it may be best included a dedicated version of the LWIP library in the repo as opposed to relying on Xilinx to supply the library. From what I have seen Xilinx does not support more than one version of LWIP at a time, and in the future it may be updated with no way to revert to the previous version.

- The LWIP RAW mode is more complicated to use than the Sockets mode but the included example projects can be used as a starting point. The biggest issue with using the RAW mode is that each SOC needs to be specifically configured in order to work properly. This is all handled by Xilinx so building off of a supplied project makes that a non-issue. The other issues is that the interrupts and timers that the protocol relies on must be controlled by the user but again the Xilinx projects handle that. What's left is to set up a series of callbacks functions that process the steps of the TCP protocol. The [LWIP wiki](https://lwip.fandom.com/wiki/Raw/TCP) has a fairly thorough breakdown of each of the steps in the TCP protocol for both the server and client, which is a good reference for determining which API functions are needed. LWIP also has an [official documentation ](https://www.nongnu.org/lwip/2_1_x/index.html) that better describes the actual API.

# Python Ethernet Client

- Python uses the `socket` library to connect to the AMDC TCP server

- The throughput I was able to achieve through the Python, as apposed to Iperf3, was significantly less. I was able to get about 200Mbps throughput using Python, however, my method of calculating throughput was likely inaccurate. I assumed a buffer size of 2MB for each transaction and measured the time to complete about 10 transactions in Python. I was able verify the size of the buffers using Wireshark. It seems likely to me that the speed is limited by the Python library and the rather slow nature of Python. I'm not quite sure how the Python library operates but it may be worth while to test if creating additional threads and connecting multiple clients to the AMDC server could speed up the Python interface. 

# SQLite Database in Python

- SQLite is very easy to set up in Python and stream data to from the TCP buffers. From what I could tell it is fast and should be able to handle the incoming data from the AMDC at high throughputs. The documentation for the Python SQLite API can be found [here](https://docs.python.org/3/library/sqlite3.html).

# Inter-core Communication

- Setting up inter-core communication on the AMDC doesn't require many changes. The architecture of the SOC is set up in such a way that both the A9 Cortex cores essentially share the same caches and ram from the users perspective. The hardware handles any issues of cache coherency or related issues so the memory can be handled as if its a single region. For ICC to work both cores need their own BSPs and their own applications. The only change that needs to be made to run two application is to change the location in memory for one core where the application binary is flashed to. This is done by modifying the linker script for one or both cores such that their memory locations do not overlap. Additionally, a shared region of memory is created that both cores will access to send data between each other.

- ICC is the only issue that is currently holding back the projects from moving forward. I was encountering an issue with debugging where memory locations from the first project were being overwritten by the second project during flashing for debugging despite the memory supposedly not overlapping in the linker script. From what I can tell this seems to be only global variables in the data section of memory that should be initialized to 0 but are not. A forum post on the [Xilinx forums](https://forums.xilinx.com/t5/Processor-System-Design-and-AXI/Running-multiple-application-files-elf-from-single-boot-bin-on/td-p/773558) about running multiple application in parallel suggests that this is an issue with C global variables that are assumed to be `0`, such as `NULL`. I have not been able to find much official documentation about running multiple applications, besides this brief doc about [linker scripts](https://www.xilinx.com/html_docs/xilinx2020_2/vitis_doc/linkerscripts.html#coo1565072993781), but from my own testing, simple applications that do not use uninitialized global variables work without issue. 
  - A possible work around may be changing `.bss` `.rodata` and `.data` sections of the first application to map to `ram` regions instead of `ddr` as that seems to prevent the global variables from being overwritten. 
  - It is possible that this is only a debugger issue so flashing the applications may prevent this issue. Additionally using a [FSBL](https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/439124055/Zynq-7000+FSBL), which I believe is needed for permanently programming the AMDC, could prevent this issue by having the FSBL load the application binary into `ddr`.

npetersen2, 2021-05-24:
Excellent Xilinx doc on Asymmetric multiprocessing (AMP) -- this is what we were trying to do above.
https://www.xilinx.com/support/documentation/application_notes/xapp1079-amp-bare-metal-cortex-a9.pdf

Provides step-by-step tutorial to get dual-cores running their own application. Includes nitty-gritty details about setting up linker scripts for memory partitioning and simple inter-processor communication.

npetersen2, 2021-10-04:
Update: upon trying to implement the app note, I discovered it is several years old and is probably out of date for the current Vivado tools...

Luckily, there is website which explains updates to the app note: https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/18842504/XAPP1079+Latest+Information

npetersen2, 2021-10-04:
The app note describes modified FSBL for AMP... 

**This is no longer needed, since later versions of the FSBL support multiple ELF files.**

npetersen2, 2021-12-21:
Closing this issue since `v1.0.x` has been released.

`v1.0.x` adds inter-core communication, Ethernet support, and real-time logging. The new code is originally based on the work that @BrandtBuchda did.

closed 2021-12-21 by npetersen2