Severson-Group/AMDC-Firmware #322: Add lwIP TCP/IP stack using FreeRTOS tasks

npetersen2, 2023-09-21:
- Replicate existing functionality in codebase, but port to FreeRTOS
- Initial goal: CLI over Ethernet (just like current codebase)

codecubepi, 2023-10-19:
Xilinx provides echo server that makes lwIP work in the context of FreeRTOS.

The AMDC has existing code to handle Ethernet. See the `while()` loop in [app_cpu0 `main.c`](https://github.com/Severson-Group/AMDC-Firmware/blob/freertos/sdk/app_cpu0/src/main.c)

I will need to find a way to merge these two implemenations, or at least verify that the Xilinx-provided implementation matches our current functionality.

codecubepi, 2023-12-12:
Investigate other FreeRTOS 'heap_X.c' implementations to see if they can be used to revamp the logging interface, per the discussion in the Dec 12 Firmware Meeting.