Severson-Group/AMDC-Firmware #340: Get Hello, World FreeRTOS program working on a Single Core

codecubepi, 2023-10-26:
Split from #328 

Modify the downloaded FreeRTOS kernel to make it compatible with the AMDC Zynq-7000 platform

- [x] Use SDK Debug tool to see if the FreeRTOS code is actually being stepped through
- [ ] Check if FreeRTOS is being given a timer
- [x] Start with only CPU0, can try to run `xil_printf()` before any FreeRTOS scheduler stuff runs
- [ ] Essentially, write "Hello, world!" program using FreeRTOS tasks that runs _on a single core_, **on hardware**

Resources:
- [FreeRTOS Reference](https://www.freertos.org/fr-content-src/uploads/2018/07/FreeRTOS_Reference_Manual_V10.0.0.pdf)
- [Xilinx Wiki - FreeRTOS](https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/18842141/FreeRTOS)
- [Zynq-7000 Overview](https://docs.xilinx.com/v/u/en-US/ds188-XA-Zynq-7000-Overview)
- [Zynq-7000 TRM](https://docs.xilinx.com/viewer/book-attachment/mxcNFn1EFZjLI1eShoEn5w/pnoMLQXFIWQ6Jhoj0BUsTQ)

codecubepi, 2023-10-26:
This was completed with the FreeRTOS-provided example application.

I am running into issues with dual-core execution: when I try to run the Hello World app on both cores, core 0 never reaches `main()` and core 1's tasks don't seem to run after being created.

### Single-core results:

[image]

[image]

npetersen2, 2023-10-26:
Excellent work! :) great to see this working on hardware. @codecubepi says the code for this is still local on this PC.

closed 2023-10-26 by npetersen2