Severson-Group/AMDC-Firmware #350: Testing multiple interrupts from FPGA, different priorities, preemption, etc

npetersen2, 2023-11-30:


annikaolson, 2024-03-01:
After looking at some online resources and discussing on the 2/27 meeting, we realized the issue was very likely dual-core related.

I tried running it on CPU 0 - with the basic interrupt that is manually set (i.e. `interrupt = slv_reg0` and have a `test_int()` function in `timing_manager.c` that sets slv_reg0 to 1) - and that called the ISR.

[image]

annikaolson, 2024-03-04:
Got it working, but I had to initialize the timing manager (and thus the PL-PS interrupts) **after** the scheduler - I put a print statement in the scheduler isr to see if that was still being called, and it is - so this might work? it might just be more efficient to put everything in one function, though

[image]