Severson-Group/AMDC-Firmware #355: Test that when you enable each sensor, we can probe with the scope to see the physical sensor is actually being driven

annikaolson, 2023-12-13:


annikaolson, 2023-12-13:
Currently, I have not gotten anything to show up on a scope. I have created a 3-bit wide debug port on the timing manager that I have hooked up to a GPIO output port.

The basic logic I am trying to implement in the AXI file:
```Verilog
always * (@posedge clk, @negedge axi rstn) begin
    if (! axi rstn)
        debug <= 0
    else if (pwm_carrier_high || pwm_carrier_low)
        debug <= ~debug
end
```

And that isn't even producing any results.

I have an event qualifier based on what the user determines the be the trigger, but I also tested everything is being written to slave registers correctly, so that isn't the issue - it's something in the FPGA.

Instead of trying to figure out why that logic isn't working, I've rebuilt the project at this point and am going to add more of my logic as I go until I get down to the issue of what isn't working - seems like the best avenue here.

I'm wondering if its the FPGA timer being gone.. thought it was unused, maybe not? I just don't know what could cause such simple logic to not work - so trying to figure that out!

npetersen2, 2023-12-14:
@annikaolson  sounds like a great plan. Hopefully by starting from a working copy, you can quickly figure out the problematic changes and get it debugged!

annikaolson, 2023-12-15:
Well, figured out the issue very early on as I was rebuilding. I mistyped the AXI reset signal that I was hooking up to my timing manger source file and for the test flop (misspelled S_AXI_ARESETN as S_AXI_RESETN) - so I bet it was inferring a wire and setting it to 0, and always resetting. :upside_down_face: :upside_down_face:

Good news is, after fixing the name, the flop was working, so I tested each of the eddy current sensors and those are working (they match the waveforms I got for the new project I made earlier). So now it’s just testing to make sure the data is all good (i.e. the reported time) and the other sensors

closed 2024-01-04 by annikaolson