Severson-Group/AMDC-Firmware #361: Update REV D block design to be compatible with v2 changes

annikaolson, 2024-02-14:
Update `rev_d.bd` such that it is compatible with the new v2 changes, notably the I/O management side.

codecubepi, 2024-06-03:
I've done a first pass at the REV E block design in commit [8f60d5c](https://github.com/Severson-Group/AMDC-Firmware/pull/397/commits/8f60d5c1bf2f9b7deea53740cb298e157c3190de). See PR #397. This is just so that @anirudhupadhyaya can get started with the pre-release `v1.3` code while @npetersen2 and I continue to debug.

Once the debugging of the following issues...

- #393 

...is complete, I will go back and update all block designs one last time:

- [ ] REV F
- [ ] REV E
- [ ] REV D
- [ ] Also, all the `hier_gpio` hierarchies still have input ports for `pwm_carrier_high` and `pwm_carrier_low`. These ports are no longer needed in the GPIO hierarchies since all the PWM syncing is handled by the timing manager now.

(This will be necessary because the debugging will likely involve another round of FPGA changes 🙃)

codecubepi, 2024-08-21:
Completed in PR #412.

closed 2024-08-21 by codecubepi