Severson-Group/AMDC-Firmware #363: Make/Find Basic Zynq-7000 Interrupt Project

codecubepi, 2024-02-27:
Learn how to use the software generated interrupts in the Generic Interrupt Controller (GIC) on the Zynq-7000, so that I can proceed on #347 

I will attempt to find/make/modify a very basic project that simply sends an interrupt from one core to another and turns on an LED or similar function.

codecubepi, 2024-03-02:
As of now (01 Mar 24, 22h45), I have modified [this example](https://github.com/Xilinx/embeddedsw/blob/2516d5ed8161e16c2813b0e8e4ceac693f23de5c/XilinxProcessorIPLib/drivers/scugic/examples/xscugic_example.c) to be dual-core compatible and have got core-to-core interrupts going from CPU0 to CPU1 (but only that direction).

Attached is a zip of the dual-core projects, since I don't want to commit this test project to any branch, but still want to have a checkpoint of a semi-working state.

[interrupt_projects.zip](https://github.com/Severson-Group/AMDC-Firmware/files/14468092/interrupt_projects.zip)

codecubepi, 2024-03-19:
As of now (18 Mar 24, 21h45), I have modified [this example](https://github.com/Xilinx/embeddedsw/blob/2516d5ed8161e16c2813b0e8e4ceac693f23de5c/XilinxProcessorIPLib/drivers/scugic/examples/xscugic_example.c) to be dual-core compatible and have got core-to-core interrupts going **in both directions**. 

The culprit seemed to be that CPU1's interface to the GIC was not being intialized, since the `CPUInitialize()` call in `XScuGic_CfgInitialize()` is gated off. So I copied the two register writes in `CPUInitialize()` into CPU1's `main()` function. That seemed to magically do the trick.

Attached is a zip of the dual-core projects, since I don't want to commit this test project to any branch, but still want save a copy of the working state.

[interrupt_proj_working.zip](https://github.com/Severson-Group/AMDC-Firmware/files/14644603/interrupt_proj_working.zip)

### Console Output

|                                                | **CPU 0** | **CPU 1** |
|------------------------------------------------|:---------:|:---------:|
| Attempt to interrupt the other core            | I         | i         |
| Interrupt written to SGI register successfully | J         | j         |
| Processed interrupt in ISR                     | Z         | z         |

[image]

npetersen2, 2024-04-02:
Nice work!

closed 2024-04-02 by npetersen2