Severson-Group/AMDC-Firmware #365: Add support in timing manager for AMDS sensor

npetersen2, 2024-03-05:
The [AMDS](https://docs.amdc.dev/accessories/amds/) system needs to be integrated into the new timing manager system.

@npetersen2 will update this issue more soon with details about how to do this.

For now, @annikaolson can read:

- [Docs about using the AMDS as an AMDC user](https://docs.amdc.dev/getting-started/user-guide/amds-interface.html)
- [Docs about how the AMDS firmware works](https://docs.amdc.dev/accessories/amds/firmware/)
- [AMDS firmware project readme](https://github.com/Severson-Group/AMDS/tree/develop/Mainboard/Firmware)

codecubepi, 2024-04-16:
Also see https://github.com/Severson-Group/AMDS/issues/47

codecubepi, 2024-05-07:
- [ ] Generate new `amdc_amds` IP
- [ ] Add `amds_X_done` ports (X is 0-3) to `amdc_timing_manager` IP
- [ ] Update REV F block design
- [ ] Update REV E block design
- [ ] Update REV D block design

closed 2024-06-20 by codecubepi