Severson-Group/AMDC-Firmware #366: Add firmware support for new AMDC REV F hardware

npetersen2, 2024-03-25:
We recently released a new AMDC hardware revision: [`REV20231005F`](https://github.com/Severson-Group/AMDC-Hardware/tree/develop/REV20231005F), see the PR: https://github.com/Severson-Group/AMDC-Hardware/pull/228.

Per the last page of the schematics, the changes for this revision are rather small:

- Add termination on GPIO differential inputs
- Update pull-down resistor value for boot-up PWM output protection
- Update P/Ns and footprints for manufacturability

These require no functional changes to the firmware---**the REV E firmware will work with the new REV F hardware!**

## Items to do

Even though users can simply use `AMDC_REV_E` firmware config to run the new REV F hardware, we should make REV F appear as "fully supported" in the docs and the firmware. This requires the following changes:

### FPGA support

- [ ] Copy the block diagram `/hw/amdc_reve.bd` to `/hw/amdc_revf.bd` and update any internal text to say "REV F" etc
- [ ] Copy the constraints file `/hw/constraints_amdc_reve.xdc` to `/hw/constraints_amdc_revf.xdc` and update text to say "REV F" etc
- [ ] Add a new `/import_revf.tcl` file which is a clone of the `import_reve.tcl` file, but with the variable `proj_revision` set to `revf`
- [ ] Run the new import script in Vivado and confirm it works seamlessly

### DSP support

- [ ] Update `drv/hardware_targets.h` to add `AMDC_REV_F` define
- [ ] Update the default hardware target `USER_CONFIG_HARDWARE_TARGET` to `AMDC_REV_F` in `usr/user_config.h`
- [ ] Search the codebase for `AMDC_REV_E` and make any `ifdef`s also apply for `AMDC_REV_F`

### Documentation support

See https://github.com/Severson-Group/docs.amdc.dev/issues/92

npetersen2, 2024-03-25:
Note that these changes are targeting the `v1` codebase and should be based off `develop`.

We will release this new REV F support as `v1.2.0`.

codecubepi, 2024-04-02:
Closed with merge of #367

closed 2024-04-02 by codecubepi