Severson-Group/AMDC-Firmware #374: Add REV F support to v1.3 release branch

npetersen2, 2024-04-23:
Since `v1.3-staging` was branched off `develop` a while back, we need to merge `develop` back into `v1.3-staging` to pull in the latest changes, i.e., REV F support, see #367 etc.

codecubepi, 2024-05-01:
Moving to Done, I believe @annikaolson and I were able to get this figured out earlier. We can review and close this issue in next week's meeting.

We rebased `v1.3-staging` off of `main`, and then created a new `user/annikaolson/timing-manager` branch off of the rebased `v1.3-staging`. A PR (#379) has been opened which will merge in the rest of Annika's commits that were on the `io-management` branch.

@npetersen2 It would be great if you could review #379 soon, so we can merge that into `v1.3-staging`. This would be nice because then Annika and I can branch off of `v1.3-staging` and continue working.

npetersen2, 2024-05-01:
Thanks for the update @codecubepi

codecubepi, 2024-05-11:
Forgot to close in our meeting on Tue May 7, doing so now.

closed 2024-05-11 by codecubepi