Severson-Group/AMDC-Firmware #380: Remove PWM Carrier High/Low Trigger logic from Eddy Current Sensor IP

codecubepi, 2024-05-07:
See the following remnant code in the Eddy Current Sensor IP's HDL:

https://github.com/Severson-Group/AMDC-Firmware/blob/d70ed83c1d96e3018c687eb137f22dfb09dba81d/ip_repo/amdc_eddy_current_sensor_2.0/hdl/amdc_eddy_current_sensor_v2_0_S00_AXI.v#L458-L473

I believe all the logic upstream of "trigger" is now taken care of by the timing manager IP, correct? I.E. The individual sensor IPs should not have to do any qualification using the PWM carrier and register configuration bits...?

It seems to me like we could remove `trigger_on_high` and `trigger_on_low` since they aren't used anymore, and the `pwm_carrier_high` and `pwm_carrier_low` input ports aren't needed anymore since they are now qualified in the timing manager, instead of in the sensor IP.

If @annikaolson can confirm this is all correct, I can make the change in my `user/codecubepi/update-amds-interface` branch, since I am already making FPGA changes. Else we can open another more-appropriately-named branch to fix this once the AMDS updates are done.

ALSO for whoever does get this done, don't forget to update the eddy current sensor IP [README](https://github.com/Severson-Group/AMDC-Firmware/blob/v1.3-staging/ip_repo/amdc_eddy_current_sensor_2.0/README.md), specifically the fact that `slv_reg3` is not longer used as `PWM_TRIGGERS` config register.

annikaolson, 2024-05-10:
If its not being used for anything else in there, then it can be removed since I have the state machine starting on my trigger signal now

codecubepi, 2024-05-14:
This issue is addressed by [db1ad1c](https://github.com/Severson-Group/AMDC-Firmware/commit/db1ad1c70b8ab54d5896605b02bfa23a8d5997cc)

closed 2024-06-20 by codecubepi