Severson-Group/AMDC-Firmware #382: Invert PWM Signals during Switching Operation

codecubepi, 2024-05-10:
Question from Aidar:

Why can't we change the PWM configuration during switching operation? He is looking to invert the driver signals to his switches during runtime, essentially switch the signals that go to the top and bottom switches of each leg of his inverter.

This seems like it could be a straight forward addition to the `amdc_inverters` IP block, add a mux that selects between the outputs of the `single_leg_switch` module for `inverter1_pwm[0]` and `inverter1_pwm[1]` ..., based on a value in a register that can be written from the C code. 

(Although it would also be good to make sure that the gate driver signals can only invert their behaviour on the high/low of the PWM carrier).

[image]

elsevers, 2024-05-15:
In the AMDC-Firmware repo, we aim to implement approaches to modulation of inverters that are useful to a typical user and strive to avoid fragmentation through inclusion of niche features that are only used in one-off cases. So far, this has meant that the firmware/hardware implement the sine-triangle PWM approach that is typical of today's motor drives. 

If users of the AMDC want to work with the AMDC firmware development team to implement a change to the core firmware which will be useful across many applications, we welcome well-documented PRs which will help make the platform more useful for everyone.

On the other hand, researchers who wish to test out non-standard approaches to motor drives to explore a new idea are encouraged to modify the inner workings of the AMDC as needed within their own repositories. We're happy to have the platform used this way and even provide resources in [docs.amdc.dev](https://docs.amdc.dev) to support this. For this particular issue, it looks like our tutorial that explains [how to create (or modify) FPGA IP cores](https://docs.amdc.dev/getting-started/advanced-tutorials/fpga-ip-core/index.html) would do the trick.

I am going to close this issue for now, but can reconsider if anyone makes a case that this is feature that will be useful to a significant number of AMDC users and is interested in contributing a PR.

codecubepi, 2024-05-22:
I'm going to switch this to "Close as not planned"

closed 2024-05-22 by codecubepi