Severson-Group/AMDC-Firmware #385: Re-order slv_regs in Timing Manager IP

codecubepi, 2024-05-21:
## Abstract

Once my FPGA changes have been merged with @annikaolson 's C code changes back into `v1.3-staging`, we should probably go in and  re-arrange the slv_regs tobe a more logical order. 

### The current order is:

| slv_reg Number | Offset | Purpose               |
|----------------|--------|-----------------------|
| 0              | 0x00   | Auto-trigger Config   |
| 1              | 0x04   | Sensor Enable Bits    |
| 2              | 0x08   | User Ratio            |
| 3              | 0x0C   | PWM Sync (High/Low)   |
| 4              | 0x10   | Unused                |
| 5              | 0x14   | Eddy Times            |
| 6              | 0x18   | Eddy Times            |
| 7              | 0x1C   | ADC and Encoder Times |
| 8              | 0x20   | ISR                   |
| 9              | 0x24   | Unused                |
| 10             | 0x28   | Trigger Timer         |
| 11             | 0x2C   | AMDS Times            |
| 12             | 0x30   | AMDS Times            |

### I think the following would make more sense:

| slv_reg Number | Offset | Purpose               |
|----------------|--------|-----------------------|
| 0              | 0x00   | Auto-trigger Config   |
| 1              | 0x04   | Sensor Enable Bits    |
| 2              | 0x08   | User Ratio            |
| 3              | 0x0C   | PWM Sync (High/Low)   |
| 4              | 0x10   | ISR                   |
| 5              | 0x14   | Trigger Time          |
| 6              | 0x18   | ADC and Encoder Times |
| 7              | 0x1C   | AMDS Times            |
| 8              | 0x20   | AMDS Times            |
| 9              | 0x24   | Eddy Times            |
| 10             | 0x28   | Eddy Times            |

This is because all the configuration registers at the beginning (lower addresses) are unlikely to be deleted or rearranged, and all of the sensor times are at the end, so that future sensors added are just added to the end of the list.

## This will require:

- [ ] Final updates to the Verilog
- [ ] Updates to the C code to make sure it agress with the FPGA (I would add defines to the `timing_manager.h` header file like so:

https://github.com/Severson-Group/AMDC-Firmware/blob/ffb82a68fb8ad68202e5c96c3ce61c1d4cb02859/sdk/app_cpu1/common/drv/amds.h#L24-L36

- [ ] and another round of hardware testing to make sure nothing is wonky.
- [ ] The README in the Timing Manager folder of the IP repo will also need to be updated.

codecubepi, 2024-05-21:
Hey @codecubepi , while you do this, you may also want to put the `enable_bits` and anything else in the timing manager IP hierarchy in the same (and following) order, for consistency:

- ADC
- Encoder (these are first since they are built-on to the base AMDC)
- AMDS 1
- AMDS 2
- AMDS 3
- AMDS 4 (since it's the most common and oldest sensor peripheral)
- Eddy 1
- Eddy 2
- Eddy 3
- Eddy 4 (latest)

and any new sensor peripherals will follow.

codecubepi, 2024-05-22:
I've addressed the FPGA and C code changes in [00b695f](https://github.com/Severson-Group/AMDC-Firmware/commit/00b695faa2133fb303fec3b7208b70126bdb131c).

https://github.com/Severson-Group/AMDC-Firmware/blob/00b695faa2133fb303fec3b7208b70126bdb131c/ip_repo/amdc_timing_manager_1.0/hdl/amdc_timing_manager_v1_0_S00_AXI.v#L527-L537

https://github.com/Severson-Group/AMDC-Firmware/blob/00b695faa2133fb303fec3b7208b70126bdb131c/sdk/app_cpu1/common/drv/timing_manager.h#L22-L52

I'll leave the updates to the README to @annikaolson

npetersen2, 2024-05-28:
@annikaolson will update the README in the upcoming PR she will make

codecubepi, 2024-05-31:
[Looks good](https://github.com/Severson-Group/AMDC-Firmware/blob/user/annikaolson/scheduler-update/ip_repo/amdc_timing_manager_1.0/README.md) in PR #392 : 

I'll close this now.

closed 2024-05-31 by codecubepi