Severson-Group/AMDC-Firmware #402: Enabling/disabling sensors during operation leads to inconsistent scheduler tick loop time

codecubepi, 2024-06-20:


codecubepi, 2024-07-29:
## Mode 0: Scheduler ISR is generated based on just the PWM carrier and User Ratio

Plots are shown for PWM 100 kHz and ratio of 10, for a control frequency of 10 kHz (expected time between Scheduler ISR calls is 100us)

[image]

[image]

[image]
 
[image]

## Mode 1: With no sensors enabled, should function just like Mode 0 above. With sensors enabled, Scheduler ISR is generated after all sensors are done

These plots are shown for again, PWM 100 kHz and ratio of 10, for a control frequency of 10 kHz (expected time between Scheduler ISR calls is 100us). These are enabling and disabling the ADC. The previous comment in the conversation shows enabling and disabling the AMDS.

[image]
[image]
[image]
[image]

npetersen2, 2024-07-29:
@codecubepi huh, interesting.

I am wondering if the `10 usec` interval when the sensors are disabled is actually just the PWM carrier period?

Can you run another test where you set the PWM carrier to 200 kHz and the ratio such that the control rate is still 10 kHz? Then, show the same plots and see if the low part of the signal goes down to 5 usec.

codecubepi, 2024-07-31:
Enabling/disabling AMDS in timing manager in all cases:

### PWM 100 kHz, Ratio 10

[image]

### PWM 200 kHz, Ratio 20

[image]

### PWM 150 kHz, Ratio 15

[image]

Wacky... so it seems that the bad scheduler frequency is not directly correlated with the PWM/ratio settings. But there is definitely something wrong in all cases only when all sensors are disabled.

Again, I wasn't able to observe this behaviour in my simulation.

codecubepi, 2024-08-05:
Progress!

I made some (_what should be_) meaningless FPGA changes that may have fixed the problem... apparently Vivado's elaboration and synthesis of combination logic controlling toggling a flop is odd.

## Changes made ([7e88b4e](https://github.com/Severson-Group/AMDC-Firmware/commit/7e88b4eedd47883ca0312278de25fba5915e7d19?diff=split&w=0#))

[image]

## Screenshots of Results

### PWM 100 kHz, Ratio 10

[image]
[image]
[image]
[image]

### PWM 200 kHz, Ratio 20

[image]
[image]

### PWM 150 kHz, Ratio 15

 
[image]
[image]

**Unfortunately:** the loop time for the control task is double what it should be, which results in the output load frequency being halved (I am seeing 300 Hz instead of 600 Hz on the uInverter/scope bench).

codecubepi, 2024-08-05:
I was able to band-aid fix the problem with the doubled loop time and halved output frequency by adding additional tolerance when deciding to schedule the task in `scheduler_run()`. This seems to be due to the floating point inaccuracies seen before. For this particular frequency set, I had to increase the `SCHEDULER_INTERVAL_TOLERANCE_USEC` from -0.06 (60 ns) to -0.15 (150 ns)... which is quite a lot.

codecubepi, 2024-08-07:
## Results for the 150 kHz PWM, Ratio 15 case, with 150ns `SCHEDULER_INTERVAL_TOLERANCE_USEC` here:

[image]
[image]

Note that this is still with no event-synchronized latching of the sensor enable signals. Also... this whole idea of enabling and disabling sensors _while the control task is running_ is pretty unlikely to come up anyway. So, good to go?

codecubepi, 2024-08-08:
See commit [8db3f4d](https://github.com/Severson-Group/AMDC-Firmware/pull/410/commits/8db3f4d7336f0a48f686dce0eef5334baae5be40), which adds a convenient way for the user to override the scheduler tolerance value if abnormal timing settings necessitate doing so.

closed 2024-08-08 by codecubepi