Severson-Group/AMDC-Firmware #404: Validate that the `v1.3` codebase still correctly logs data for non-integer sampling rates

npetersen2, 2024-06-26:
Per this discussion: https://github.com/Severson-Group/AMDC-Firmware/pull/401#discussion_r1648072962

The new `v1.3` codebase easily allows non-integer sampling rates for the control task. We have updated the scheduler to handle this by keeping track of time using `double` type variables. This seems to be working well.

However, the logging system still uses `uint32_t` types to keep track of time. I have a feeling this will lead to logging issues for some configurations of system sample rates.

Before we release `v1.3`, we need to make a comprehensive test suite to verify that logging actually works for different settings. For example, set up a sinusoid at 1/10 the control rate in the code, then log it and check the output data still looks as it should. Do this for a range of sample rate settings and log settings.

codecubepi, 2024-07-16:
Hey @npetersen2 and @codecubepi of the future!

For the `v2` codebase, to avoid weird time checks in the logging system, the `v2` logging system will not support logging at a given _frequency_, only a ratio of the samples (every sample, every other, every third, etc) much like how the timing manager runs the scheduler at a ratio of the PWM carrier.

codecubepi, 2024-07-18:
Closing with merge of #408

closed 2024-07-18 by codecubepi