Severson-Group/AMDC-Firmware #424: Setting timing manager event ratio to 1 increases Run Time significantly in some cases

Known4225, 2024-11-11:
There's some sort of issue with the timing manager (or perhaps statistics reporting) that causes the runtime of a task to increase significantly with certain timing manager + pwm frequency configurations.

This was tested on the controller task from the [VSI tutorial](https://docs.amdc.dev/getting-started/tutorials/vsi/index.html).

The runtime for this task should be around 3.5us, but in some cases it is much higher (as demonstrated in [this PR thread](https://github.com/Severson-Group/docs.amdc.dev/pull/112#discussion_r1835540636))

I did additional testing with other PWM frequencies to see where this bug occurs (with an event ratio of 1).

Tested PWM frequencies:
250kHz - Run Mean is 22.8us
110kHz - Run Mean is 16.6us
105kHz - Run Mean is 10.2us
100kHz- Run Mean is 10.1us
95kHz - Run Mean is 10.1us
94kHz - Run Mean is 10.1us
93kHz - Run Mean is 8.4us
92kHz - Run Mean is 3.5us
91kHz - Run Mean is 3.5us
50kHz - Run Mean is 3.5us

Additional testing has confirmed that if `PWM Frequency` / `Timing manager event ratio` > `46kHz`, then this bug occurs, and the severity of it appears to be linear with the `PWM Frequency`. (That is, having 250kHz PWM and 2 as timing manager event ratio yields a Run Mean of 22.8us still, because 250 / 2 > 46, and the timing manager event ratio has no effect on the linear scaling of this bug).

Actually, it is more like 22.4us with an event ratio of 2 (and 250kHz). Not sure what to make of any of this information.

elsevers, 2024-11-14:
Thanks @codecubepi, @Known4225. 

We would like to be able to operate this with control rates in the 100 kHz range which would imply event ratios of 1.

Nathan actually did a fun demo with a control rate of 200 kHz (PWM of 1 MHz) in [this paper](https://ieeexplore.ieee.org/abstract/document/10238914). Would that combination of frequencies (event ratio of 5) have this problem?

We also should support Si IGBTs, which typically have switching frequencies between 5 kHz and 20 kHz, and so these would be apt to result in event ratios of 1. I see from your notes above, that this is likely okay, but can you confirm?

@Known4225, could you do some more debugging on this? 

@codecubepi, do you have suggestions on what he can check to identify the root cause of this issue? (@Known4225 says he'll check the loop time)

Known4225, 2024-11-14:
I've compiled about 50 data points changing the PWM frequency and the TASK_UPDATES_PER_SEC. All of these experiements used a timing manager event ratio of 1. The results are strange (formatting is run-time/loop-time).

https://docs.google.com/spreadsheets/d/13YE25-4grKAyvpcY-a8ns06K56rwdKYc6dkgXZXi2XQ/edit?usp=sharing

Also I did a separate test with 1000kHz PWM, the control task at 200kHz, and a timing manager ratio of 5. The result is a run-time of 10.08us and a loop-time of 22.00us (it should be loop-time of 5us, so it isn't working).