Severson-Group/AMDC-Firmware #435: Get AMDC CAN interface working and provide demonstration of successful communication with 3rd party component

elsevers, 2025-01-22:
# Goal
Get AMDC CAN interface working and provide demonstration of successful communication with 3rd party component

# Task List 
See subissues

elsevers, 2025-01-22:
Notes from our 1/22 meeting:

- @jimhgant and @elsevers have assembled CAN information and the eLev lab's CAN interests on this issue: [here](https://github.com/Severson-Group/DOE_GHG_Simulations/issues/102) @mohamed-dek1 and @Known4225 should read it.
- We want to come up with a plan for what CAN support the AMDC should have. Should we be adding a J1939 module to the AMDC? 
- End of semester goal:  @mohamed-dek1 should demonstrate communication with a BorgWarner `HVLP-20` motor drive in ME 76. See [this comment](https://github.com/Severson-Group/DOE_GHG_Simulations/issues/102#issuecomment-2606079598) for details on the drive
- We will use the `AMDC-firmware` repo and `AMDC firmware team` project board to track progress and tasks on this goal.
- @mohamed-dek1 and @Known4225 should create a Gantt chart in @mohamed-dek1's Friday slides for this project and then set up the first necessary GitHub issues. Link the issues as sub-issues to this issue.

jimhgant, 2025-01-22:
I made a [new folder in our shared drive](https://drive.google.com/open?id=1d28-isbYmEo5e2cFF1YBFluJ6euolkyz&usp=drive_fs) that contains a product sheet, a more detailed product manual, and a decoder file (.dbc) for the HVLP-20 motor drive we will use to test AMDC CAN. If/when more information is collected I will place it in this folder.

mohamed-dek1, 2025-05-07:
05/07/2025

# AMDC CAN Report