Severson-Group/AMDC-Firmware #437: Write CAN driver on AMDC.

mohamed-dek1, 2025-01-30:


mohamed-dek1, 2025-02-14:
- The current BSP does not have the required CAN files that are referenced in the existing header files.
- Need to figure out how to configure or add the necessary files to begin testing the previous implementation.

mohamed-dek1, 2025-02-19:
`can init` and `can deinit` seem to be working well but some of the other commands like `can selftest` are returning an error.

### Error message
[image]

mohamed-dek1, 2025-02-26:
The current configuration of the GPIO ports on the AMDC does not seem to support the CAN board. I will have to figure out where the configuration is written and update it to allow communication to a CAN board properly.

Known4225, 2025-02-28:
02/28/25:
GPIO mux needs expansion to add CAN device. GPIO mux system needs documentation, comments, and changing the names of the enum values in gpio_mux_device_t to reflect the name of the device that each entry corresponds to.

mohamed-dek1, 2025-03-21:
CAN is going through the GPIO but not at the expected pins.
Will remap those in vivado to match the expected pins on the CAN board

mohamed-dek1, 2025-04-25:
4/25/25:

attempted to set up AMDC to AMDC CAN communication.

It did not work. Here are the results:

We have a strange CAN "error" that causes the CAN signal to be corrupted in the following manner:

GOOD SIGNAL:

[image]

BAD SIGNAL:

[image]

We can get the GOOD SIGNAL by routing the CAN from CANA to CANB on a single CAN board.

[image]

However, we get the BAD SIGNAL in any of the following configurations

[image]
[image]
[image]