Severson-Group/AMDC-Firmware #448: AMDS Data Error?

codecubepi, 2025-02-01:
I think there may be an error with how the FPGA layer of the AMDS driver handles incoming data from the AMDS mainboard.

The [Texas Instruments ADS8860](https://www.ti.com/lit/ds/symlink/ads8860.pdf) analog-to-digital converter on the sensors cards requires that the analog voltage measured is prescaled to a 0.0 - $$V_{REF}$$ range (we use 5V for $$V_{REF}$$, see sensor card schematics [here](https://github.com/Severson-Group/AMDS/blob/v2.0.x/HighVoltageCard/REV20200913C/Schematic.pdf) in the [AMDS](https://github.com/Severson-Group/AMDS) repo). The ADC then outputs over SPI (to the AMDS board's STM MCU) an unsigned 16-bit number. The following chart shows how the card input voltage is prescaled for the ADC and then mapped to a 16-bit number (see page 21 of the ADC datasheet to verify).

| **Card Input Voltage** | **ADC Analog Input Voltage** | **ADC Output (16-bit unsigned)** |
|------------------------|------------------------------|----------------------------------|
| 500V                   | 5V ($$V_{REF}$$)             | 0xFFFF                           |
| 250V                   | 3.75V                        | 0xC000                           |
| 0V                     | 2.5V                         | 0x8000                           |
| -250V                  | 1.25V                        | 0x4000                           |
| -500V                  | 0V                           | 0x0000                           |

This 16-bit unsigned number is then sent from the AMDS STM MCU to its AMDC FPGA driver over UART as described [here](https://docs.amdc.dev/accessories/amds/firmware/index.html). It is at this point I believe the error occurs. The FPGA performs a *sign-extension* to 32-bits when placing the unsigned value into the output register, instead of a *zero-extension*. 

https://github.com/Severson-Group/AMDC-Firmware/blob/03bc92a1773b2a35ea688f3fd763f48a2849b461/ip_repo/amdc_amds_1.0/hdl/amdc_amds_v1_0_S00_AXI.v#L756-L767

The C code layer of the driver (amds_get_data()) then reads the full *32-bit signed* value. This will cause any value over the middle of the range (unsigned numbers with a leading 1) to be treated incorrectly treated as negative, and the negative values (which in unsigned have a leading 0) to be treated as positive.

https://github.com/Severson-Group/AMDC-Firmware/blob/03bc92a1773b2a35ea688f3fd763f48a2849b461/sdk/app_cpu1/common/drv/amds.c#L56-L71

If all this is the case, I'm not sure how things have ever worked. As far as I can tell, this sign-extension has been performed since the first release revision of the "motherboard" driver back in AMDC-Firmware v1.0.0:

https://github.com/Severson-Group/AMDC-Firmware/blob/b02347a3d921e78221e3f8130cb0efd4628da4c5/ip_repo/amdc_motherboard_1.0/hdl/amdc_motherboard_v1_0_S00_AXI.v#L693-L701

When I was updating the "motherboard" driver to the Timing Manager-compatible "AMDS" driver in Summer 2024, I did not catch this issue as I was more concerned with the timing of sampling and data transmission, not with the data itself. So the issue persists into the current release (v1.4.x).

This issue was found by @psflann while prepping for the ECE 504 Drive Lab course at UW-Madison.

npetersen2, 2025-02-01:
See #426

codecubepi, 2025-02-01:
Ah, thanks @npetersen2 . I figured this had to be a known issue, so I searched the existing issues for "AMDS" and came up short. #426 did not appear because it's a discussion... whoops on my part 🤷‍♂️

Known4225, 2025-02-06:
I'm attempting to recreate this bug before applying the suggested fix.

I'm not able to recreate using the Low Voltage card

[image]

Top is voltage according to ADC, bottom is raw value from AMDS.

Here's the result when the "quick fix" is applied (taking the raw value from the AMDS and doing `& 0xFFFF` to it.

[image]

The schematic uses a slightly different ADC part, the [ADS8865IDGSR](https://www.ti.com/lit/ds/symlink/ads8865.pdf?HQS=dis-dk-null-digikeymode-dsf-pf-null-wwe&ts=1738883982259&ref_url=https%253A%252F%252Fwww.ti.com%252Fgeneral%252Fdocs%252Fsuppproductinfo.tsp%253FdistId%253D10%2526gotoUrl%253Dhttps%253A%252F%252Fwww.ti.com%252Flit%252Fgpn%252Fads8865), perhaps this could be why?

Does this bug only occur on the high voltage card and current card?

codecubepi, 2025-02-07:
Hmm yeah it looks like the differing ADC parts may be an issue.

For the current and HV cards, reference page 21 of the [ADS8860 datasheet](https://www.ti.com/lit/ds/symlink/ads8860.pdf) for the ADC transfer function. That ADC outputs a 16-bit unsigned number.

For the LV card, reference page 20 of the [ADS8865 datasheet](https://www.ti.com/lit/ds/symlink/ads8865.pdf). That ADC outputs a 16-bit signed value.

This means we can't treat all data coming from the AMDS the same. Somewhere in the chain, either on the AMDS or AMDC (probably the AMDC), the user should be able to tell the `amds_get_data()` function what card they are reading from, and then the function will read the 16-bit number accordingly.

Known4225, 2025-04-18:
This has been completed by #449

PR #449 (2025-04-14): Fix AMDS data error
This PR closes #448.

It adds helper functions to the AMDS driver file that
- allow users to read in AMDS measurements with the correct sign
- factor in the specifics of the AMDS sensor cards to correctly interpret the digital values as an ADC voltage
- provide support to convert ADC voltage measurements to the sensed signals along with default gain and offset values

Future work should update docs.amdc.dev to inform users about these functions.
closed 2025-04-18 by Known4225