Severson-Group/AMDC-Firmware #455: Ensure functionality of CAN board (power, ground, ICs)

mohamed-dek1, 2025-03-23:


mohamed-dek1, 2025-04-22:
CAN board SN:01 has functional electrical characteristics.

mohamed-dek1, 2025-04-29:
SN:02 voltages at CAN_H and CAN_L in idle (relative to CAN GND).

CAN_H: 2.5V
CAN_L: 2.5V

These characteristics are identical to can board SN:01

mohamed-dek1, 2025-04-29:
We were able to communicate through CAN AMDC to AMDC with the following configuration:

[image]

[image]

It seems like CAN is only working right now when we connect CAN0 to CAN1 on a single board. A CAN signal sent from CAN0 to CAN1 within a board will be successfully transmitted to the CAN bus and any other device on this bus can also receive this signal.

Our setup in this image is connecting CAN0A, CAN1A, and CAN0B (where A and B denote different CAN boards).

CANXA is connected to AMDCA, and CANXB is connected to AMDCB

So we do have working AMDC to AMDC CAN communication, but only under these special conditions.

Directly connecting CAN0A to CAN0B results in the CAN signal being "corrupted"

[image]

The exact cause of this is unknown, but the signal is "corrupted" unless CAN0 is connected to CAN1 on any board (the effect is instant, disconnecting CAN0 from CAN1 while CAN is being sent immediately corrupts the signal, and reconnecting them immediately restores it)

elsevers, 2025-05-01:
Nice progress!

mohamed-dek1, 2025-05-07:
Both boards have been individually tested by looping CANA and CANB peripherals together and verifying the signal going through the bus.

closed 2025-05-07 by mohamed-dek1